Mechanical-Advantage/AdvantageScope #341: Running from Source - Out of Memory Error

DeanDavis, 2025-03-05:
**Describe the bug**
Forked the repo and installed prerequisites on a Mac (M1 chip).
First bug was an exception when running "npm run watch"
"xrClientBundles not defined"
Looked at the rollup config and line 154 was trying to return xrClientBundles but, yes, it was not defined.
xrBundles was so I changed like 154 to just xrBundles.

Reran and things looked better then I got (this is just the portion of terminal output that contains the error)
------
bundles src/sourceListHelp.ts → bundles/sourceListHelp.js...
created bundles/sourceListHelp.js in 8.3s
bundles src/betaWelcome.ts → bundles/betaWelcome.js...

<--- Last few GCs --->

[43438:0x120008000]    59776 ms: Mark-Compact 4032.9 (4132.1) -> 4024.1 (4139.1) MB, pooled: 0 MB, 2293.83 / 0.08 ms  (average mu = 0.410, current mu = 0.141) allocation failure; scavenge might not succeed
[43438:0x120008000]    63810 ms: Mark-Compact 4040.7 (4139.3) -> 4031.1 (4145.6) MB, pooled: 0 MB, 3969.00 / 0.12 ms  (average mu = 0.192, current mu = 0.016) allocation failure; scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Reached heap limit Allocation failed - JavaScript heap out of memory
----- Native stack trace -----

 1: 0x1029831e4 node::OOMErrorHandler(char const*, v8::OOMDetails const&) [/usr/local/bin/node]
 2: 0x102b82fd8 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, v8::OOMDetails const&) [/usr/local/bin/node]

----------

**To Reproduce**
Download repo, try to run on Mac

**Expected behavior**
a running version of AdvantageScope

jwbonner, 2025-03-08:
The issue with `xrClientBundles` seems to be a separate issue (just a typo in the default configuration). AdvantageScope has a large number of bundles, so the configuration is set up so you can run `watch` on them separately to reduce memory usage. For example, to run `watch` on the primary bundles used in the main renderer you can run:

```
nom run watch -- --configLargeRenderers
```

There are other bundle sets that you can find [here](https://github.com/Mechanical-Advantage/AdvantageScope/blob/c10831bf11e20ec491b1d566f7700a4ba049dbbf/rollup.config.mjs#L154-L159) if they're relevant to your use case.

godmar, 2025-10-02:
The issue seems to be due to large memory usage rather than a memory leak. Therefore, we can work around by simply increasing node.js's heap size to 16GB as in:
```bash
NODE_OPTIONS="--max-old-space-size=16384" npm run watch

....
created bundles/shared$loadRobot.js in 11.7s
bundles src/xrClient/xrClient.ts → bundles/xrClient.js...
created bundles/xrClient.js in 10.7s
bundles src/xrControls.ts → bundles/xrControls.js...
created bundles/xrControls.js in 4.2s

[2025-10-02 09:36:11] waiting for changes...
```

blaze-developer, 2026-01-30:
I have consistently encoutered the same issue, although I find it odd that some people dont experience this. Do some platforms have different heap size limits by default?