Mechanical-Advantage/AdvantageScope #342: Advantage Scope Line Graph negative times

reslip1, 2025-03-06:
**Describe the bug**
My limelight will sometimes log a negative time in wpilog network tables. This causes the log to start at -4000 seconds, then have 30-60 seconds of data from my test. This makes visualizing data near impossible. 

**To Reproduce**
Steps to reproduce the behavior:
Load log with negative timestamps and positive timestamps. Open WPIlog. See that data that is negative can't be filtered and creates difficulty viewing

**Expected behavior**
Ignore any negative timestamps as bogus data

**Screenshots**

[image]

**Version (required):**
Version: 4.1.2
Distributor: WPILib
Platform: win32-x64
Build Date: 2/1/2025, 24:07:35 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
Add any other context about the problem here.

jwbonner, 2025-03-08:
In general, we try to avoid modifying data that is part of the original log file. Can you elaborate on how this "makes visualizing data near impossible"? It seems like you should just be able to zoom in as described in the [documentation](https://docs.advantagescope.org/tab-reference/line-graph#viewer-pane) to see the part of the log that's relevant to you.

One option would be setting the initial zoom to not include negative values, but I also think there's value in conveying to the user that the log may include data they weren't expecting (especially when it's easy to just zoom in, and AdvantageScope preserving the time range you were previously viewing when switching log files).

jwbonner, 2025-03-08:
I pushed a change so that the default visible range will not include negative values, so you need to specifically zoom out to see those values. I think this is probably the most reasonable behavior since it doesn't remove any information from the log but makes the default view reasonable for most use cases.

closed 2025-03-08 by jwbonner