Mechanical-Advantage/AdvantageScope #344: Cannot read PhotonPipelineResult in logs

Gold856, 2025-03-06:
**Describe the bug**
When opening a wpilog, NT:/photonvision/Camera Name/rawBytes only shows the raw byte array.

**To Reproduce**
Steps to reproduce the behavior:
1. Open [FRC_20250305_141246.log](https://github.com/user-attachments/files/19102447/FRC_20250305_141246.log) (file extension changed to allow GitHub upload) in ascope.
2. Go to NT:/photonvision/Camera Name/rawBytes.
3. I only see rawBytes, and I can drag it and view the raw byte array.

**Expected behavior**
 I should see a dropdown that allows me to inspect the the contents of rawBytes.

**Screenshots**
[image]

**Version (required):**
Version: 4.1.2
Distributor: WPILib
Platform: win32-x64
Build Date: 2/1/2025, 24:07:35 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
I can access the decoded contents of rawBytes when live viewing over NT.

Gold872, 2025-03-06:
Just to add to this, one thing I noticed is that when I'm connected live to NT, I'm able to view rawBytes entry just fine.

jwbonner, 2025-03-08:
Looks like we somehow missed adding support for Photon structs from log files at all, so it was only supported for NT. At least it was an easy fix...

closed 2025-03-08 by jwbonner