Mechanical-Advantage/AdvantageScope #345: Empty Metadata despite having entries

blackgnezdo, 2025-03-09:
**Describe the question you have.**

I'm following [these instructions](https://docs.advantagescope.org/tab-reference/metadata/) to capture the git build info as Metadata. The code is here:
https://github.com/team6962/Code-2025/blob/45e17644e91661a61cd2fad004b5ac874068197e/src/main/java/frc/robot/RobotContainer.java#L271-L276

**Describe the reason for your confusion.**

Once the robot runs, I see the relevant output in both the console and the keys of my wpilog file:
```
% strings ./logs/FRC_20250309_000942.wpilog | grep git
/Metadata/git.commit.message.short
/Metadata/git.commit.user.email
/Metadata/git.commit.user.name
/Metadata/git.commit.id.abbrev
/Metadata/git.branch
/Metadata/git.build.host
/Metadata/git.commit.time
/Metadata/git.dirty
git.commit.message.short: Make sure jar task depends on generateGitProperties
git.commit.user.email: blackgnezdo@gmail.com
git.commit.user.name: Greg Steuck
git.commit.id.abbrev: 45e1764
git.branch: main
git.build.host: abhobe-laptop
git.commit.time: 2025-03-08 12:51:04
git.dirty: true 
```

Yet, when I open the file in AdvantageScope, the Metadata tab doesn't show any of the logged keys.

abhobe, 2025-03-10:
[image]

this is what it looks like in sim and on the roborio

jwbonner, 2025-03-10:
Can you upload the log file where you see this issue?

abhobe, 2025-03-10:
https://drive.google.com/file/d/1hgrBUuFaf6t0frxlzkx36N_L0ZKbrk8G/view?usp=sharing

lmk if the link is broken

jwbonner, 2025-03-10:
It looks like the screenshot you posted before is from a live connection. Since you're using a `StringLogEntry` instead of a `StringPublisher`, the metadata will be saved to the DataLog but not published to NetworkTables. It looks like the metadata appears correctly when opening the log file you sent in AdvantageScope (screenshot below). Are you having trouble seeing the metadata in the log or only when connected over NetworkTables?

[image]

blackgnezdo, 2025-03-10:
Thanks @jwbonner. We mostly care about looking at this info in the logs, so it's good enough. @abhobe could you confirm that it works for you this way?