Mechanical-Advantage/AdvantageScope #346: Log timestamp alignment doesn't work when opening multiple logs at once

jamesdooley4, 2025-03-11:
**Describe the bug**
Time synchronization only appears to work if a wpilog file is opened first, 

**To Reproduce**
Steps to reproduce the behavior:

1. File -> Open Logs..
2. Select three attached logs
3. See error

or

1. Open Advantage Scope
2. Drag and drop the three attached logs
3. See error

or

1. File -> Open Logs...
2. Select two of the three logs (doesn't matter which, tried with both wpilog + hoot, or two hoot logs)
3. File -> Add New Logs...
4. Select the third log file
5. See error

**Expected behavior**
The enabled time should be aligned, regardless of how the log files were opened.

**Screenshots**
Correct behavior, when opening one log file first (tried with both wpilog & either hoot file), then adding the two other logs:
[image]

Incorrect behavior, when opening all three at once or opening two logs and then adding the third one:
[image]

**Version (required):**
Version: 4.1.4
Distributor: FRC 6328
Platform: win32-x64
Build Date: 3/8/2025, 20:53:35 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
Could consider this an [enhancement] since it does require the app picking which of the multiple initial log files is the time base to use.

Log files:
[WASAM_E3_69E1E2A34C32435320202059260117FF_2025-03-09_12-52-44.zip](https://github.com/user-attachments/files/19177359/WASAM_E3_69E1E2A34C32435320202059260117FF_2025-03-09_12-52-44.zip)

[WASAM_E3_rio_2025-03-09_12-52-44.zip](https://github.com/user-attachments/files/19177375/WASAM_E3_rio_2025-03-09_12-52-44.zip)

[FRC_20250309_195252_WASAM_E3.zip](https://github.com/user-attachments/files/19177381/FRC_20250309_195252_WASAM_E3.zip)

jwbonner, 2025-03-11:
Thanks for catching that. It looks like there was an edge case regarding some variants of the "enabled" key. The fix will be included in the next release, or you can download a development build [here](https://github.com/Mechanical-Advantage/AdvantageScope/actions/runs/13798679704) in the meantime.

jlmcmchl, 2025-03-25:
I'm still able to replicate this bug with v4.1.5 - sample trio of logs at this link: https://drive.google.com/drive/folders/144Yd9pdBZgojVvN7tayh0BS23nSZ4MLp?usp=sharing

[image]

jwbonner, 2025-03-26:
Are you

jwbonner, 2025-03-29:
I pushed a change (9c45801215b0c8c76ca04e52eb29704825a9f4ba) which seems to fix the issue by always prioritizing the `DS:enabled` field over `FMSControlData` if the former is available. The log files you send sync up properly with that change since the data from `DS:enabled` is valid:

[image]

closed 2025-03-29 by jwbonner