Mechanical-Advantage/AdvantageScope #348: Trouble rendering my robot's 3D model into Advantage Scope

Mukesh-Kalikaya, 2025-03-20:
**Describe the bug**
Cannot import my robot model into Advantage Scope.

**To Reproduce**
1. I followed the documentation and made config.json and model.glb and model_0.glb.
2. I connected to the simulation.
3. I dragged the robot pose into 3D field Poses and tried to select the robot but couldn't see it.

**Expected behavior**
I expected to see my robot name in the list but couldn't.

**Screenshots**
[image]
[image]
[image]

**Version (required):**
Version: 4.1.2
Distributor: WPILib
Platform: win32-x64
Build Date: 2/1/2025, 24:07:35 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

**Additional context**
Add any other context about the problem here.

Mukesh-Kalikaya, 2025-03-20:
```
{
    "name": "Eclipse",
    "rotations": [],
    "position": [
        0,
        0,
        0
    ],
    "cameras": [],
    "components": [
        {
            "zeroedRotations": [],
            "zeroedPosition": [
                0,
                0,
                0
            ]
        }
    ]
}
```

Here is my config.json file. Forgot to include it earlier.

jwbonner, 2025-03-20:
It looks like when selecting the custom assets folder, you chose the “Robot_Eclipse” folder. Try selecting the “userAssets” folder instead (the one that _contains_ “Robot_Eclipse”).

Mukesh-Kalikaya, 2025-03-20:
That did fix it. Thank you 👍.

closed 2025-03-20 by Mukesh-Kalikaya