Mechanical-Advantage/AdvantageScope #349: Pose for robots is always 0,0,0 for my custom VisionOS app based on AdvantageScopeXR

toddyocum, 2025-03-22:
I was trying to implement a native VisionOS version of AdvantageScopeXR, so based on RealityKit, and have made decent progress, but am not seeing the robots positions changing from 0,0,0, even though in the AdvantageScopeXR client they are, there the robot moves the same as it does on the 3d field in AdvantageScope. 

So trying to figure out if my assumption was wrong about how the AdvantageScopeXR client works, I assumed that the ThreeDimensionRendererCommand sent as part of XRPacket would have robot position updates? But it appears not, so before debugging further, thought I'd check to see if my assumption is wrong. 

In my swift code, I am successfully decoding the messages from the XRServer, and am seeing this in my debug console for ThreeDimensionRendererCommand  XRPacket (3322 2025 is the custom robot asset) 

Received command with 1 objects
Processing command for game: 2025 Field (Welded) with 1 objects
Processing command for robot : 3322 2025 with pose: 0.000000, 0.000000,  0.000000

Any insights is greatly appreciated. Thanks for sharing such a great project!

[image]

jwbonner, 2025-03-25:
Yes, the `ThreeDimensionRendererCommand` contains the current poses of all objects on the field. If the data is appearing correctly in the official AdvantageScope XR app, that suggests that there is an issue with the way your client is decoding the data.

As an aside, I just want to be transparent that it is *unlikely* that we would be interested in adopting and maintaining an official version of AdvantageScope XR based on RealityKit. The current app is specifically designed to share as much code as possible with the normal 3D field (including downloading it from the server to ensure compatibility); we frequently make changes to the object types and visualization code, and it's just not practical for us to maintain multiple copies of that code. You are absolutely welcome to create tools like this for your own (or others’ use), it's just unlikely to be something we'll be able to take on officially. We're happy to talk further if you have questions or concerns about this.

toddyocum, 2025-03-28:
Thanks for the feedback, that was very helpful! I think you are right, decoding the data is an issue. I assumed because my app was decoding the other messages from the XRServer, that everything would decode ok. With some digging, that appears to not be the case for the position. I'm digging into it now. 

And totally get the not including any of this. It is considerate of you to take the time to mention it. I would have offered it for inclusion if there was interest, but am fine with it not being included. This is just a side project of personal interest for me.

closed 2025-03-28 by toddyocum