Mechanical-Advantage/AdvantageScope #351: 3D field over video view

reecelikesramen, 2025-03-23:
**Is your feature request related to a problem? Please describe.**
Feature is not related to a problem or issue.

**Describe the solution you'd like**
Use AprilTag detection to overlay the field/robot/3D field viz elements over a video recording of a match. Estimate camera position and FOV then overlay a translucent field with a 3D view replicating the camera over the video.

**Additional context**
My team plans on experimenting with this during the offseason this year. Thought it would be interesting/useful to compare inline the difference between robot perspective and reality.

jwbonner, 2025-03-25:
Can you elaborate on what you would imagine this to look like? I'm not quite following how AprilTag detection is relevant (is AdvantageScope detecting AprilTags?). It seems like there are several things very close to this that are already possible in AdvantageScope:

1. You can display a video alongside the 3D field (including custom camera placements and FOVs):

[image]

2. You can also use the [fixed camera view](https://docs.advantagescope.org/tab-reference/3d-field#fixed-camera) to recreate the view from an onboard camera, as seen [here](https://youtu.be/buXa1xXWU_U?t=865). If you recorded video from onboard the robot, it could be overlayed similar to the above example.

It seems like you're imagining a translucent video such that both views would be directly overlayed on top of each other, but I'm a little doubtful of whether that would make things easier to visualize rather than just adding visual noise. I'm curious if there's something else you have in mind though.

reecelikesramen, 2025-03-25:
I wrote this in kind of a haste at the airport yesterday, so I'll elaborate on some things:
- We want to visualize more clearly the difference between what the robot is doing on the field versus the robot's actual state. Comparing a video and a field side-by-side is very helpful, but showing them overlayed could help pinpoint issues in localization or odometry.
- I imagine that AdvantageScope is detecting the AprilTags in the video to extract the position and FOV that the video was taken frame. The idea is that only chosen 3D field elements (like the robot model, swerve states, vision targets) would overlay on top of the video. I've only worked on small projects with AprilTags in OpenCV, so I'm not sure how feasible this is with and without camera matrices and distortion coefficients.

If you think this is outside of the scope of AdvantageScope, this is certainly respectable. Either way, I wanted to put this feature suggestion out in the public to see if other users would benefit from a visualization like this.

blaze-developer, 2025-03-31:
This kind of thing is possible, but detecting apriltags to extract position and fov seems like a huge task. If such a thing cannot be done, maybe the user can align it themselves? You should be able to figure out the FOV of a match recording, and just align from there.

j5155, 2025-08-26:
For this feature to be useful, it would be really nice to customize the transparency of each element independently so that just the recorded robot could be overlaid without the field (for example).
Another way to implement alignment that might be easier is some way of letting the user mark points on the video that correspond to points on the 3d field render, and have the render automatically zoom to align them. You could get really fancy and track those points to compensate for camera motion, but even one alignment would make comparisons much easier.