Mechanical-Advantage/AdvantageScope #355: Video download and auto synchronization do not work

katzuv, 2025-03-28:
**Describe the bug**
Downloading a match video from TBA and YouTube does not work. Both emit the same error:
[image]
Also, even after the video has been loaded, auto synchronization doesn't work.
**To Reproduce**
[Example log file](https://github.com/Galaxia5987/robot-2025/blob/c46ad7222754e6e401fdaca98654a06c9248e587/logs/DCMP/akit_25-03-26_14-09-32_iscmp_q40.wpilog) from ISR DCMP Q40. [TBA link](https://www.thebluealliance.com/match/2025iscmp_qm40)

**Expected behavior**
Video should be loaded from TBA or YouTube buttons, and synchronize automatically.

**Version (required):**
```
Version: 4.1.5
Distributor: FRC 6328
Platform: win32-x64
Build Date: 3/18/2025, 04:19:02 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0
```
**Additional context**
Add any other context about the problem here.

jwbonner, 2025-03-29:
I pushed a couple of changes (3fa1d4051159f780e53be21bc35b106f34916ac7 and 8c047d586a41c6657fb09359184cdeb8a42b7832) which should address the issue of synchronization with a local video files. As described in the documentation, downloading from YouTube or TBA may break due to changes outside of our control. We will continue to monitor for updates to the relevant AdvantageScope dependencies, but otherwise any issues with those features are out of our control.

jwbonner, 2025-03-30:
I pushed another fix (3261be089da14eb5faa7f00912fbc56a4c7fb404) which seems to fix the TBA/YouTube download feature, though as always it may break in the future because YouTube frequently makes server-side changes that affect it.

closed 2025-03-30 by jwbonner

katzuv, 2025-03-30:
Thank you so much!