Mechanical-Advantage/AdvantageScope #359: Add an option to view discrete fields in other tabs

katzuv, 2025-03-28:
**Is your feature request related to a problem? Please describe.**
In the competition, we compared a lot of times the pose of the robot to whether it's aligning or not (which is a boolean trigger we log). It'd be nice to be able to see the trigger state on the Odometry or 3D Field tabs directly.
**Describe the solution you'd like**
Add an option to see discrete fields (and maybe line graphs too) in other tabs.

jwbonner, 2025-03-28:
AdvantageScope already supports overlaying multiple tabs using the window pop-out functionality. We frequently use that feature for exactly this use case. It seems like this is a more flexible solution to the problem. Is there something you feel is missing from it?

katzuv, 2025-03-28:
I was imagining something like this (sorry for the bad paint edit):
[image]
I guess the graph is too cramped but single (boolean and non-boolean values) might fit nicely. But I do see the point about the pop-out, we don't use it enough.

blaze-developer, 2025-03-29:
> AdvantageScope already supports overlaying multiple tabs using the window pop-out functionality. We frequently use that feature for exactly this use case. It seems like this is a more flexible solution to the problem. Is there something you feel is missing from it?

Something that can be missing with that option is that often users have a 1080p screen where splitting the screen across multiple windows makes it hard to see with the windows being so small and skinny. This feature could make this easier for some users, but it could also be implemented as the ability to open discrete Fields in a right sidebar on any tab, or the ability to open a window dedicated only to discrete fields.

ArchdukeTim, 2025-04-19:
I would like the discrete fields to be similar to how they are on the line graph view (at the bottom), but on the odometry tab specifically