Mechanical-Advantage/AdvantageScope #365: NT:/FMSInfo/FMSControlData prevents DS:enabled from being read in getEnabledData()

jonahsnider, 2025-03-28:
**Describe the bug**

`NT:/FMSInfo/FMSControlData` is being read even when it has no data, which prevents `getEnabledData()` from properly outputting when the robot is enabled, even when `DS:enabled` and `DS:autonomous` are present.

**To Reproduce**

Steps to reproduce the behavior:

[log.wpilog.txt](https://github.com/user-attachments/files/19512550/log.wpilog.txt)

1. Download this log, remove `.txt` extension
2. Open file in AdvantageScope
3. Observe that there is no autonomous or enabled indicator in odometry views

**Expected behavior**

The provided log file should properly show the timespans where the robot is enabled.

**Screenshots**

N/A

**Version (required):**

```
Version: Development
Distributor: FRC 6328
Platform: darwin-arm64
Build Date: 3/28/2025, 21:08:59 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0
```

**Additional context**

If you comment out the element for `NT:/FMSInfo/FMSControlData` in `LogUtil.ts`, the bug goes away

jwbonner, 2025-03-28:
Looks like the same root cause as the followup to #346. I've opened https://github.com/wpilibsuite/allwpilib/issues/7885 to hopefully address the root cause, though implementing a workaround in AdvantageScope should be relatively simple in the meantime.

jonahsnider, 2025-03-28:
Looks like the issue is related to `DataLogManager` capturing all NT traffic by default, including FMSInfo. If you configure it to stop after the logger has started, the entry will just be frozen in time. This is often when the robot is disabled.

jwbonner, 2025-03-28:
Properly supporting "finish" records essentially means dealing with `null` values for fields and/or having the set of fields change over time. We have no plans to support either of those features (they existed in part in previous versions, and were a *massive* maintenance burden). In general, we also want to avoid tying AdvantageScope's internal data representations to WPILOG/NT. There are other fixes related to the handling of "finish" records which we are currently looking at, but I think the best solution in this case is to simply rely on the `DS` table when available.

jwbonner, 2025-03-29:
I pushed a couple of changes which should fix the problem:

- 9c45801215b0c8c76ca04e52eb29704825a9f4ba applies consistent priorities when searching for known fields, so `DS:enabled` is always preferred over `FMSControlData` when available (same thing for autonomous state, match data, etc).
- f3d55eaf89bf8842dacbbbd01b3c365d65f5a55e ignores any WPILOG entries that are finished within a second of being started. This keeps the `NT` table out of the sidebar entirely, which I think is appropriate since it doesn't contain meaningful data.

closed 2025-03-29 by jwbonner

jonahsnider, 2025-03-29:
Awesome, thank you so much for the quick fix!