Mechanical-Advantage/AdvantageScope #374: Migration to BSD license

jwbonner, 2025-04-29:
Since its initial release, AdvantageScope has been distributed under the MIT license available [here](https://github.com/Mechanical-Advantage/AdvantageScope/blob/main/LICENSE). With anticipated changes to AdvantageScope to support the 2027 control system, we are proposing a migration to the BSD 3-clause license shown below:

```
Copyright (c) 2021-2025 Littleton Robotics. All rights reserved.

Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions are met:

- Redistributions of source code must retain the above copyright
  notice, this list of conditions and the following disclaimer.
- Redistributions in binary form must reproduce the above copyright
  notice, this list of conditions and the following disclaimer in the
  documentation and/or other materials provided with the distribution.
- Neither the name of Littleton Robotics, FRC 6328 ("Mechanical Advantage"),
  AdvantageScope, nor the names of other AdvantageScope contributors may be
  used to endorse or promote products derived from this software without
  specific prior written permission.

THIS SOFTWARE IS PROVIDED BY LITTLETON ROBOTICS AND OTHER ADVANTAGESCOPE
CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT
NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY NONINFRINGEMENT
AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL
LITTLETON ROBOTICS OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT
NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY
OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE,
EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
```

The permissions granted by the new BSD license are very similar to AdvantageScope’s current license, with the sole exception of a restriction that products derived from AdvantageScope refrain from using the name “AdvantageScope” (or the other names listed in the license text).

AdvantageScope is currently in the process of adding support for FTC with the 2027 control system, which will significantly expand the number of people using and modifying AdvantageScope. 6328 highly encourages this type of work; team-driven modifications of AdvantageScope have significant benefits for student learning, and may lead to upstream contributions that benefit all AdvantageScope users. However, we acknowledge the potential for user confusion arising from the distribution of multiple versions of “AdvantageScope.” To address this concern, this change will clarify that “AdvantageScope” refers only to the primary, officially supported version of the application. We believe this modification is crucial to support AdvantageScope users during and after the upcoming control system changes.

<details>
<summary>What about older versions of AdvantageScope?</summary>
It’s true that any version of AdvantageScope released prior to this change will still be available under an MIT license. Switching to the BSD license for future releases covers the most common case of a user forking the latest version of AdvantageScope, while also making the wishes of the developers more clear regarding use of the name “AdvantageScope”. By the time the first stable release of AdvantageScope is available for FTC, the last MIT-licensed version will be over 2.5 years old; we do not expect that users will want to fork a version of AdvantageScope that is significantly outdated.
</details>

<details>
<summary>Are you trying to discourage forks of AdvantageScope?</summary>
Not at all! We encourage users to experiment with modifying AdvantageScope, and to contribute changes back to the original so that all users can benefit from the work of the community. However, we think it is important that users (especially those new to FIRST) can easily understand which versions of the software are officially supported. By making our position on the use of the “AdvantageScope” name more clear, we hope to make it <i>easier</i> for teams to develop their own modifications without causing any inadvertent harm to the rest of the software ecosystem.
</details>

We are looking for feedback on this change from all previous contributors to AdvantageScope. If your username is listed below, you have contributed code that is part of the current version of AdvantageScope (thank you!). We are open to any feedback, suggestions, or questions you might have; you can contact us by replying to this issue or emailing [software@team6328.org](mailto:software@team6328.org).

**If you agree that your contributions can be released under the proposed BSD license above, all you need to do is react to this issue with 👍.**

Thank you for your cooperation and contributions to AdvantageScope!

- [x] @jwbonner
- [x] @brettbbonner
- [ ] @jonahsnider
- [ ] @mcm001
- [ ] @AlphaPranav9102
- [ ] @Tuna156
- [ ] @srimanachanta
- [ ] @Gold872
- [ ] @17acres
- [ ] @chauser
- [ ] @spacey-sooty
- [ ] @auscompgeek
- [ ] @m10653
- [ ] @bznerd
- [ ] @person4268

Tuna156, 2025-04-29:
Good luck. Is there a list of things that you need to move on over to FTC? I probably won't be able to help since I'm only part of FRC but I thought I should still ask.

jwbonner, 2025-04-29:
@Tuna156 Thanks for asking! Over the next couple of months we'll be working to put together a better public roadmap regarding changes for 2026 (FRC) and 2027 (FRC + FTC), so keep an eye out for that. We're also putting together some clearer contributions guidelines which should clarify what changes/features are most likely to be approved.

And regardless of the specific work for 2027, there are always a ton of feature on the [backlog](https://github.com/Mechanical-Advantage/AdvantageScope/issues?q=is%3Aissue%20state%3Aopen%20label%3Aenhancement) (both big and small). A lot of my attention for the upcoming offseason will be focused on 2027, so we will gladly accept help from the community if there are any of those feature requests that catch your eye. Let us know if you have any questions!

Tuna156, 2025-04-29:
My current focus is on custom asset models, regarding optimizations, and hopefully streamlining that process with a setup wizard for it (I believe I saw an issues talking about that). I think it would be helpful. Is that something you would likely accept?

jwbonner, 2025-04-29:
Beyond 2027 changes, a setup wizard for robot models (#292) is actually the #1 feature that we'd like to add before 2026! As you may have guessed, custom robot assets are still the most pain point we hear from teams; anything to streamline that process would be much appreciated. If you have specific ideas about how to approach it or what that wizard should look like, I'd also be happy to take a look and give suggestions before you get too far down any particular path.

person4268, 2025-04-29:
While I don't intend to, will there be a way to change the branding of AdvantageScope to suit the new naming requirements? Or is the expectation something different?

jwbonner, 2025-04-29:
@person4268 The new license restricts the use of the name "AdvantageScope" to "endorse or promote products" derived from the original, so the expectation would be that a user who wants to distribute their own version of AdvantageScope would change the branding. I'm not sure there's a feasible way to make the name *easily* configurable, but it would certainly be doable if someone wanted to take that approach. Of course, in the majority of cases we would prefer that someone developing changes to AdvantageScope simply contribute them back to the original project and avoid the issue entirely :)

jwbonner, 2025-05-12:
Thanks you to everyone that has given their approval for relicensing! We currently have approval for >99.9% of contributions as measured by LOC, and are following the example set by Mozilla that [at least 95% approval](https://blogs.fsfe.org/ciaran/?p=58) is sufficient to move forward with relicensing. If you have any additional questions or concerns, please reach out to software@team6328.org.

PR #383 (2025-05-12): Switch to BSD license & add license headers
Closes #374
closed 2025-05-12 by jwbonner