Mechanical-Advantage/AdvantageScope #382: Seperate Video Tab Window Has Graphical Glitches

blaze-developer, 2025-05-09:
When opening a video to synchronize to a log file, it is necessary to open other visualization windows alongside it. However, in all of my experiences, across both Linux and Windows machines, opening the Video tab in its own window results in graphical bugs. Examples include jittering video frames, videos disappearing for periods of time, trouble moving the window around, and the window jumping around in position.

Video Here:
https://files.catbox.moe/e2p0ee.mp4

jwbonner, 2025-05-09:
Do you see this on any of the other popup windows with a fixed aspect ratio (2D field, swerve, etc)? All of the issues you show in the video seem to be related to resizing or moving the window, so it would be helpful to establish whether this is related to the video tab in particular or the fixed aspect ratio.

blaze-developer, 2025-05-09:
These issues also appear when just leaving the window alone after being resized. This issue does not seem to arise for the other tabs, although seemingly there is a difference in the way that the aspect ratio is handled. On the other tabs, like odometry, when resizing to something with an incorrect aspect ratio, the window is fine with there being some spaces on the edges. However, the video tab, after being resized, seemingly auto corrects itself and resizes itself to perfectly fit.

https://files.catbox.moe/jtyqn3.mp4