Mechanical-Advantage/AdvantageScope #392: Menu bar flydowns not showing anything.

ThadHouse, 2025-06-30:
**Describe the bug**
None of the menu bar items are showing anything, except for Help, which is definitely missing a few things.

**To Reproduce**
Open AdvantageScope 2027 Alpha 1. Try to click any of the menu bar items.

**Expected behavior**
Menu bar items work.

**Screenshots**

[image]

**Version (required):**
About menu not able to be shown, but 2027 alpha 1.

**Additional context**
Windows 64 bit.

ThadHouse, 2025-07-12:
I need to reopen this. Its _better_, but a lot are still missing.

ThadHouse, 2025-07-12:
[image]

jwbonner, 2025-07-12:
What’s missing from the menu? That screenshot looks right from what I remember.

ThadHouse, 2025-07-12:
Oh. Things moved. Thats what was confusing me. Nevermind. I was thinking that was the file menu.

closed 2025-07-12 by ThadHouse

jwbonner, 2025-07-12:
Yeah, a bunch of those options used to be under “Help” which didn’t really make sense.