Mechanical-Advantage/AdvantageScope #397: High CPU/Memory usage on specific log file

gerth2, 2025-07-05:
**Describe the bug**
When loading a 50Mb log file in table view in the 2027 alpha release, memory usage goes up to 4Gb+ and the UI becomes unresponsive.

[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:

[sketchy file download here
](https://limewire.com/d/wOcPS#7K364jvKZR)

**Expected behavior**
less RAM usage and a moving UI

**Screenshots**
see above

**Version (required):**

Version: 4.1.5
Distributor: WPILib
Platform: win32-x64
Build Date: 3/18/2025, 04:21:56 UTC
Electron: 32.0.2
Chromium: 128.0.6613.84
Node: 20.16.0

jwbonner, 2025-07-05:
The issue is that AdvantageScope is trying to render the raw values in hexadecimal, which isn't very practical for values that are  almost a megabyte each. I implemented one potential solution, which is to just cut off values larger than 0.5 Kb (and arrays longer than 500 elements). The performance and memory usage of this seems to be much more reasonable.

[image]

jwbonner, 2025-07-18:
The fix for this is now released for the 2027 alpha (and will be available in the 2026 beta).

closed 2025-07-18 by jwbonner