Mechanical-Advantage/AdvantageScope #422: + doesn't work

spacey-sooty, 2025-09-05:
**Describe the bug**
I click the + on the top right corner and don't get a drop down menu to add a new tab.

**To Reproduce**
Steps to reproduce the behavior:

1. Open AdvantageScope
2. Click the +

**Expected behavior**
The drop down to add a new tab appears

**Screenshots**
https://github.com/user-attachments/assets/41005722-5931-4f60-a513-5212d78f51a0

**Version (required):**
Version: 26.0.0-beta-1
Distribution: FRC 6328
Platform: linux-x64
Build Date: 9/5/2025, 03:32:10 UTC
Electron: 37.2.6
Chromium: 138.0.7204.185
Node: 22.17.1

**Additional context**
Confirmed working on latest stable version

jwbonner, 2025-09-05:
I've tested this on Ubuntu 22.04 with both X11 and Wayland (as well as Windows and macOS), but I'm not able to recreate the issue. Can you try running from the command line and check if anything is printed in the console when clicking that button?

jwbonner, 2025-09-06:
Can you see if this happens on [v27.0.0-alpha-3](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v27.0.0-alpha-3)? That release uses the same version of Electron as v4.1.6 but with some other changes related to the new tab button.

spacey-sooty, 2025-09-07:
> I've tested this on Ubuntu 22.04 with both X11 and Wayland (as well as Windows and macOS), but I'm not able to recreate the issue. Can you try running from the command line and check if anything is printed in the console when clicking that button? You could also check the console in the developer tools, though it's less likely that there would be an issue in the renderer.

Nothing in the console or dev tools. I will note I'm on Ubuntu 24.04 not 22.04

spacey-sooty, 2025-09-15:
```
{
  x: 1117.4180908203125,
  y: 10.993863105773926,
  width: 27.993284225463867,
  height: 27.993284225463867
}
{
  x: 1117.4180908203125,
  y: 10.993863105773926,
  width: 27.993284225463867,
  height: 27.993284225463867
}
```
Here's the output, the menu does work on that build

spacey-sooty, 2025-09-16:
Yep that works!