Mechanical-Advantage/AdvantageScope #426: AdvantageScope will not launch under wayland.

blaze-developer, 2025-09-26:
When running the beta on my main desktop machine running under wayland, it crashes during startup.

Running it from the command line through the Wayland display server yields [these errors](https://pastebin.com/hx50z7Ad). I have up to date GPU drivers and AdvantageScope still works flawlessly through Wayland in previous versions.

When running through X11, it runs as expected.

Although this wont run on my main machine, it works as expected on my laptop, also running Wayland. This would normally lead to me to believe i have a problem with my system configuration, however previous AdvantageScope versions still run as expected.

Any help interpreting these errors would be greatly appreciated! Thanks

blaze-developer, 2026-06-28:
Are you using the AppImage? The AppImage always runs under XWayland. Try using your native version, and passing it "--ozone-platform-hint=wayland". [It was rolled back to this version because of visual bugs, so the selection of wayland is no longer automatic](https://github.com/Mechanical-Advantage/AdvantageScope/issues/77#issuecomment-3721265823), but it seems to me that it doesn't work with Wayland at all. And I am not seeing any issues on the latest electron version. Could you see if the latest electron version has any visual bugs?

blaze-developer, 2026-10-25:
I am still seeing this issue in the lastest releases. I don't have any visual bugs with the regular non "-linux" build natively on wayland, and the released versions currently still won't open a window under wayland, only under a forced xwayland session.

nullcubee, 2026-10-25:
Using the `v26.0.2` flatpak, on a NixOS host, I can reproduce advantagescope failing to open when run with the `--ozone-platform-hint=wayland` flag, which forces running via wayland, resulting in these errors: https://gist.github.com/nullcubee/86a7d332c4e0db4405d59536f19eb6db.

Running without the flag opens and runs, but via Xwayland.

However, the Xwayland version isn't using my GPU for rendering and has the following errors: https://gist.github.com/nullcubee/9eb15d4cdaf11f068a97971b9485b4fb